Apollon77/I2CSoilMoistureSensor #1: temperature sensor could read negative values...

krikk, 2015-11-14:
according to the schematic the i2cSoilMoistureSensor uses an NCP18XH103F03RB (http://www.mouser.at/ProductDetail/Murata-Electronics/NCP18XH103F03RB/?qs=k3rQb2PKXqkF9YvF09qsvw%3D%3D) for temperature measurement, this sensor can measure values down to -40 °C according to the datasheet... with my raspberry pi example code negative values are reported correctly (see https://github.com/Miceuz/i2c-moisture-sensor#raspberry-pi-example) ...perhaps you could adapt your library to also allow negativ temp. values..

thx

...i would have submitted a patch, but my arduino programming skills are not that good... (still learning the basics)

Apollon77, 2015-11-14:
Yes ... did not tested the sensor in negative areas :-) Will need to do.
And I change temperature to a signed value

Apollon77, 2015-11-14:
Have already changed it locally and will test by tomorrow evening ... stay tuned for first update and first official release :-))

Apollon77, 2015-11-15:
see 1.0.0 :-))

closed 2015-11-15 by Apollon77