Apollon77/I2CSoilMoistureSensor #4: Use with ESP8266?

theopensourcerer, 2016-05-25:
I'd like to use your library on the ESP8266. I have the ReadSensorData example compiling and running on the micro-controller but I can't see where you configure which GPIO pins to use for SCL and SDA? 

Sorry if I am being dense.

theopensourcerer, 2016-05-25:
Sorry - just found that you add the SDA and SCL to the Wire.begin(SDA, SCL) call.

closed 2016-05-25 by theopensourcerer

Apollon77, 2016-05-25:
The available WIre Libraries are used. There is a mapping in the I2CSoilMoistureSensor.cpp File rather on top for Tiny and Standard Arduino.
On this place a mapping for an ESP8266 could be added to use the Wire library available for that platform OR because it is also called "Wire" for ESP8266 as I seen it it could work stright forward.

This works because the "Wire.begin" is called in the scetch outside of the library and as I've seen you specify the pins there. So try to add the parameters as of here https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#i2c-wire-library and see if it works ... else we would need to do such a mapping.

Apollon77, 2016-05-25:
Great to have it proofed so far :-)

theopensourcerer, 2016-05-25:
Wow - thanks for the reply but I closed it before I had seen your reply lol. 

I closed it just because I read that I can call Wire.begin(0,1)... However - I am not getting any sense out of my sensors. I have switched to the other example code where it looks for an I2C device but nothing is ever found :-(