Apollon77/I2CSoilMoistureSensor #9: Cant Compile for Wemos D1 Mini (ESP 8266)

SphaeroX, 2017-04-10:
I includes the _I2CSoilMoistureSensor.h_ correctly but when i try to init it with **I2CSoilMoistureSensor sensor**; i got some compile Errors.

i also got the _ArduinoJson.h_ the compiler says:

> Arduino: 1.6.12 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"
> 
> sketch\wemos.ino.cpp.o: In function `ArduinoJson::Internals::JsonParser<ArduinoJson::Internals::StdStringTraits<String>::Reader, ArduinoJson::StaticJsonBufferBase&>::parseObject()':
> 
> sketch\include/ArduinoJson/Deserialization/JsonParserImpl.hpp:131: undefined reference to `I2CSoilMoistureSensor::I2CSoilMoistureSensor(unsigned char)'
> 
> sketch\wemos.ino.cpp.o: In function `__static_initialization_and_destruction_0':
> 
> sketch\include/ArduinoJson/Data/List.hpp:67: undefined reference to `I2CSoilMoistureSensor::I2CSoilMoistureSensor(unsigned char)'
> 
> collect2.exe: error: ld returned 1 exit status
> 
> exit status 1
> Fehler beim Kompilieren für das Board WeMos D1 R2 & mini.
> 
> Dieser Bericht wäre detaillierter, wenn die Option
> "Ausführliche Ausgabe während der Kompilierung"
> in Datei -> Voreinstellungen aktiviert wäre.
>

Apollon77, 2017-04-10:
CVan you compile when you only use the I2CSoilMoisture library? Just question to figure out if it is an interoperability issue or whatever.

In general you are the only/first one having problems compiling

SphaeroX, 2017-04-10:
Heres the standalone compile

> 
> Arduino: 1.6.12 (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"
> 
> sketch\new.ino.cpp.o:(.text.setup+0x18): undefined reference to `I2CSoilMoistureSensor::begin(bool)'
> 
> sketch\new.ino.cpp.o:(.text.setup+0x1c): undefined reference to `I2CSoilMoistureSensor::getAddress()'
> 
> sketch\new.ino.cpp.o:(.text.setup+0x20): undefined reference to `I2CSoilMoistureSensor::getVersion()'
> 
> sketch\new.ino.cpp.o: In function `setup':
> 
> C:\Users\MGasc\Desktop\new/new.ino:6: undefined reference to `I2CSoilMoistureSensor::begin(bool)'
> 
> sketch\new.ino.cpp.o: In function `HardwareSerial::begin(unsigned long)':
> 
> C:\Users\MGasc\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/HardwareSerial.h:75: undefined reference to `I2CSoilMoistureSensor::getAddress()'
> 
> sketch\new.ino.cpp.o: In function `setup':
> 
> C:\Users\MGasc\Desktop\new/new.ino:11: undefined reference to `I2CSoilMoistureSensor::getVersion()'
> 
> C:\Users\MGasc\Desktop\new/new.ino:15: undefined reference to `I2CSoilMoistureSensor::isBusy()'
> 
> C:\Users\MGasc\Desktop\new/new.ino:15: undefined reference to `I2CSoilMoistureSensor::getCapacitance()'
> 
> C:\Users\MGasc\Desktop\new/new.ino:15: undefined reference to `I2CSoilMoistureSensor::getTemperature()'
> 
> C:\Users\MGasc\Desktop\new/new.ino:16: undefined reference to `I2CSoilMoistureSensor::getLight(bool)'
> 
> C:\Users\MGasc\Desktop\new/new.ino:17: undefined reference to `I2CSoilMoistureSensor::sleep()'
> 
> sketch\new.ino.cpp.o:(.text.loop+0x48): undefined reference to `I2CSoilMoistureSensor::isBusy()'
> 
> sketch\new.ino.cpp.o: In function `loop':
> 
> C:\Users\MGasc\Desktop\new/new.ino:20: undefined reference to `I2CSoilMoistureSensor::getCapacitance()'
> 
> C:\Users\MGasc\Desktop\new/new.ino:21: undefined reference to `I2CSoilMoistureSensor::getTemperature()'
> 
> C:\Users\MGasc\Desktop\new/new.ino:24: undefined reference to `I2CSoilMoistureSensor::getLight(bool)'
> 
> C:\Users\MGasc\Desktop\new/new.ino:24: undefined reference to `I2CSoilMoistureSensor::sleep()'
> 
> C:\Users\MGasc\Desktop\new/new.ino:24: undefined reference to `I2CSoilMoistureSensor::I2CSoilMoistureSensor(unsigned char)'
> 
> sketch\new.ino.cpp.o: In function `__static_initialization_and_destruction_0':
> 
> C:\Users\MGasc\Desktop\new/new.ino:26: undefined reference to `I2CSoilMoistureSensor::I2CSoilMoistureSensor(unsigned char)'
> 
> collect2.exe: error: ld returned 1 exit status
> 
> exit status 1
> Fehler beim Kompilieren für das Board WeMos D1 R2 & mini.
> 
> Dieser Bericht wäre detaillierter, wenn die Option
> "Ausführliche Ausgabe während der Kompilierung"
> in Datei -> Voreinstellungen aktiviert wäre.
>

SphaeroX, 2017-04-10:
And i include the file with this:

#include "include/I2CSoilMoistureSensor.h"

and got the .h and .cpp at the folder

Apollon77, 2017-04-10:
Seems to me as that the .cpp file is not correctly found/referenced ...
while googling around I saw things like:

 the libs needs to be in folger  "libraries" ... ?! maybe "include" as forder is wrong?

krikk, 2017-04-12:
have running the library on the same platform without any problems...

the include is wrong, you do not include the path!! ...can be closed...

Apollon77, 2017-04-12:
thank you for your support!

closed 2017-04-12 by Apollon77