Apollon77/I2CSoilMoistureSensor #10: Not Getting Soil moisture value kindly help...

varunhr, 2017-04-18:
Hi i recently bought a chirp I2C soil moisture sensor V2.7.4 and i hooked it up to Arduino Uno Like this
Uno                          Chirp
5V                            VCC
Gnd                          GND
SDA                          SDA(10k Pullup)
SCL                           SCL(10k Pullup)

And i flashed the example code to read the the sensor value, but to my vain im only getting some junk values. This is the output of my serial port
---------------------------------------------
I2C Soil Moisture Sensor Address: 20
Sensor Firmware version: FF

Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 
----------------------------------------------

Would you Kindly let me know the problem might be?

Regards
Varun

mohaoahmadi, 2017-05-01:
you might have the wrong wiring colors(by the maker himself). The one I got was:
RED: Vcc, Blue: Ground??? , Yellow: SCL, White: SCA. 
Hope that helps.

mhaas, 2017-06-29:
Hey there,

are you using the chirp (uses a battery + speaker) or really just the cheaper moisture sensor? The chirp requires a longer delay when reading a value.

FWIW, I reformatted the two code examples given by the wemakethings page and the only real difference is the delay:

````
--- <unnamed>
+++ <unnamed>
@@ -3,6 +3,7 @@
 void setup() {
   Wire.begin();
   Serial.begin(9600);
+  writeI2CRegister8bit(0x20, 6); //reset
 }
 
 void writeI2CRegister8bit(int addr, int value) {
@@ -15,7 +16,7 @@
   Wire.beginTransmission(addr);
   Wire.write(reg);
   Wire.endTransmission();
-  delay(1100);
+  delay(20);
   Wire.requestFrom(addr, 2);
   unsigned int t = Wire.read() << 8;
   t = t | Wire.read();
@@ -24,9 +25,9 @@
 
 void loop() {
   Serial.print(readI2CRegister16bit(0x20, 0)); //read capacitance register
+  Serial.print(", ");
   writeI2CRegister8bit(0x20, 3); //request light measurement 
-  delay(9000);                   //this can take a while
+  Serial.println(readI2CRegister16bit(0x20, 4)); //read light register
   Serial.print(", ");
-  Serial.println(readI2CRegister16bit(0x20, 4)); //read light register
-  delay(500);
+  Serial.print(readI2CRegister16bit(0x20, 5)); //temperature register
 }
````

Apollon77, 2017-06-30:
because the library is designed for the normal sensors and not chirp I would not like to increase the times  ... But when this solution works fr you everythign is great ...

mhaas, 2017-06-30:
@apollon77 I don't know if the delay in question fixes OPs issue. In any case, the delay could be made configurable. I may get around to actually using my sensors in a couple of weeks, so I might submit a PR!

Apollon77, 2017-06-30:
you're welcome to :-)

gluckgabor, 2017-11-01:
Hi, I have exactly the same problem. Could You please post the solution? Increasing the delay time does not seem to solve it.

srsbh, 2017-11-13:
I had a similar problem, what worked for me is this. As I understand, every sensor has a different address. First find the address, by using the change address code and in continuation run the code for getting values of moisture. Its important that you join both codes, as the I2C does not have memory, so running them one after the other won't help. But if you find the address, and change the code to suit it, then the moisture values will start  recording.
Hope this helps.

gluckgabor, 2017-11-13:
Thanks, sure that must had been my problem too.

braghettos, 2018-02-27:
Hi, I have a different problem: when I connect the YELLOW to SCL and GREEN to SDA as pictured in the Arduino Uno Rev3 Pinout
[image], the ReadSensorData example simply doesn't show anything. It's somehow stucked.

When I remove just the YELLOW pin, the serial monitor shows just these rows:
```
I2C Soil Moisture Sensor Address: 20
Sensor Firmware version: FF

Soil Moisture Capacitance: 
```

When I remove the YELLOW pin and the GREEN pin, the serial monitor shows just these rows:

```
I2C Soil Moisture Sensor Address: 20
Sensor Firmware version: FF

Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
```

Based on the Arduino pinout above and the RED+BLACK+GREEN+YELLOW cables on the moisture sensor, what are the right connections?

gluckgabor, 2018-02-28:
Also I would be curious of the intervals you get in dry air and in cup full of water just above the sign showing "put this part in water" all under water. Raw readings you get should be around 350 at dry air and around 680-700 when sensor is completely under water.

braghettos, 2018-02-28:
Hi @gluckgabor , using only the ChangeSensorI2CAddress.ino example I have the following issues:

YELLOW pin connected to SCL, GREEN pin connected to SDA --> Nothing shown in the serial monitor

Just the GREEN pin connected to SDA --> The sketch stops after the following rows:

```
I2C Soil Moisture Sensor Address: 20
Sensor Firmware version: FF

Change address to 0x21 ...
```

Just the YELLOW pin connected to SCL --> The sketch stops after the first row:

`I2C Soil Moisture Sensor Address`

Without YELLOW and GREEN pins conneted --> 

```
I2C Soil Moisture Sensor Address: 20
Sensor Firmware version: FF

Change address to 0x21 ...... ERROR

Scanning...
No I2C devices found
```

braghettos, 2018-02-28:
Hi @gluckgabor , this is the merging I did:
```

#include <I2CSoilMoistureSensor.h>
#include <Wire.h>

I2CSoilMoistureSensor sensor(0x20);

void setup() {
  Wire.begin();
  Serial.begin(9600);

  sensor.begin(); // reset sensor
  delay(1000); // give some time to boot up
  Serial.print("I2C Soil Moisture Sensor Address: ");
  Serial.println(sensor.getAddress(),HEX);
  Serial.print("Sensor Firmware version: ");
  Serial.println(sensor.getVersion(),HEX);
  Serial.println();

}

/*loop scans I2C bus and displays foud addresses*/
void loop() {
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) {
      Serial.print("Unknow error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

  delay(5000);           // wait 5 seconds for next scan

  while (sensor.isBusy()) delay(50); // available since FW 2.3
  Serial.print("Soil Moisture Capacitance: ");
  Serial.print(sensor.getCapacitance()); //read capacitance register
  Serial.print(", Temperature: ");
  Serial.print(sensor.getTemperature()/(float)10); //temperature register
  Serial.print(", Light: ");
  Serial.println(sensor.getLight(true)); //request light measurement, wait and read light register
  sensor.sleep(); // available since FW 2.3

  delay(5000);           // wait 5 seconds for next scan
}
```
This is the result when everything is connected:

> I2C Soil Moisture Sensor Address: 20
> Sensor Firmware version: FF
> 
> Scanning...
> No I2C devices found
> 
> Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
>

gluckgabor, 2018-03-01:
Have you copied the setaddress part? Can't see it. G

Diego Braga <notifications@github.com> (időpont: 2018. febr. 28., Sze,
19:47) ezt írta:

> Hi @gluckgabor <https://github.com/gluckgabor> , this is the merging I
> did:
>
>
> #include <I2CSoilMoistureSensor.h>
> #include <Wire.h>
>
> I2CSoilMoistureSensor sensor(0x20);
>
> void setup() { [...]
-- 

Üdvözlettel, Glück Gábor
+36-20-327-1897
www.gluckgabor.hu

<gaborgluck@hotmail.com>

Apollon77, 2018-06-04:
no progress, will close. Please comment for reopen

closed 2018-06-04 by Apollon77

nachoco, 2020-05-23:
Having the same problem and cannot find a solution anywhere...

gluckgabor, 2020-05-24:
First and foremost try changing the USB cable. If replacement does not solve the problem, try checking for I2C address with I2C scanner. Set the I2C address correctly as otherwise measurement won't work.