Apollon77/I2CSoilMoistureSensor #11: Reading of getCapacitance() gets "historical data"

bilbolodz, 2017-08-02:
First reading of getCapacitance() provide "old data". To get current moisture reading you have to call getCapacitance() twice.

Apollon77, 2017-08-07:
I did not experienced this. Do you have waited a bit after starting the sensor? Such a behavious is only known for light measurements as I remember

bilbolodz, 2017-08-07:
It's battery powered sensor so after reading I'm putting it into sleep mode.

Apollon77, 2017-08-07:
Can you please check if your problem is a duplicate to https://github.com/Apollon77/I2CSoilMoistureSensor/issues/8 and solutions there work too?

bilbolodz, 2017-08-07:
I will try it this evening.

bilbolodz, 2017-08-07:
No It's not the same issue. It's not connected with sleeping/not sleeping. Solution "reading sensor version" also not working. My code is:

`
void loop()
{
[..]
  while (sensor.isBusy()) wait(50);
  float tempC = sensor.getTemperature() / (float)10;

#if defined MY_DEBUG1
  Serial.print(F("Temp C: "));
  Serial.println(tempC);
#endif

  int sensorValueM;

  //First reading gets old value so we need to read twice
  while (sensor.isBusy()) wait(50);
  sensorValueM = sensor.getCapacitance();

  while (sensor.isBusy()) wait(50);
  sensorValueM = sensor.getCapacitance();

  sensor.sleep();

  int moistureLevel = min(map(sensorValueM, min_hum_adc, max_hum_adc, 0, 100), 100);

  moistureLevel = max(0, moistureLevel);

#if defined MY_DEBUG1
  Serial.print(F("Hum i2c: "));
  Serial.print(sensorValueM);
  Serial.print(F(" Humidity: "));
  Serial.print(moistureLevel);
  Serial.println(F(" %"));
#endif
[..]
`

Apollon77, 2017-08-07:
@Miceuz: doyou have an idea?!

bilbolodz, 2017-08-07:
It could be some interaction with Mysensors code

Miceuz, 2017-08-08:
Yes, this is true. When you are reading the sensor, you are getting data from the last read. If you are doing rare reads, you should do two reads. 

This is done to avoid clock stretching issues with a lot of the controllers. Clock stretching is very poorly supported over a wide variety of controllers - Raspberry Pi, ESP8266 to name a few important ones. That's why sensor returns a previous reading and makes another one after each call.

bilbolodz, 2017-08-08:
So "it's feature not a bug"? It's battery powered sleeping sensor so reads are indeed rare.

Miceuz, 2017-08-08:
Well, it's neither feature, nor a bug. A workaround for buggy world.

Apollon77, 2017-08-08:
;-)) Thank you for all your support! I will close this one.

closed 2017-08-08 by Apollon77

bilbolodz, 2017-08-08:
> I should add this to the documentation.
Very good idea