Apollon77/I2CSoilMoistureSensor #12: Issue getting readings for multiple sensors

danielfppps, 2017-08-09:
Hi,

I have two moisture sensors that I want to read with the same arduino UNO R3. I have changed the address of one of them with the setAddress command to 0x21 (confirm with the i2c scanner this is the address) and have tried the following code:

```
//Libraries
#include <I2CSoilMoistureSensor.h>
#include <Wire.h>                     

I2CSoilMoistureSensor moisture_sensor_1(0x20);
I2CSoilMoistureSensor moisture_sensor_2(0x21);

void setup()
{
    Serial.begin(9600);
    Wire.begin(); 
    moisture_sensor_1.begin();
    delay(1000);
    moisture_sensor_2.begin();
    delay(1000);
}

void loop()
{

  float moisture_1, moisture_2;
  
  while (moisture_sensor_1.isBusy()) delay(50);
  moisture_1 = moisture_sensor_1.getCapacitance();    
  while (moisture_sensor_2.isBusy()) delay(50);  
  moisture_2 =  moisture_sensor_2.getCapacitance();    
  Serial.println(moisture_1);
  Serial.println(moisture_2);     
}
```
I also tried using a single class instance and using the changeSensor function:

```
//Libraries
#include <I2CSoilMoistureSensor.h>
#include <Wire.h>                     

I2CSoilMoistureSensor moisture_sensor(0x20);

void setup()
{
    Serial.begin(9600);
    Wire.begin(); 
    moisture_sensor.begin();
    delay(1000);
}

void loop()
{

  float moisture_1, moisture_2;
  
  while (moisture_sensor.isBusy()) delay(50);
  moisture_1 = moisture_sensor.getCapacitance();

  moisture_sensor.changeSensor(0x21, true);
  
  while (moisture_sensor.isBusy()) delay(50);  
  moisture_2 =  moisture_sensor.getCapacitance();  

  moisture_sensor.changeSensor(0x20, true);
  
  Serial.println(moisture_1);
  Serial.println(moisture_2);     
}
```
However all I am getting from the sensors is -1 for both moisture and temperature. Both sensors read correctly if I plug only one sensor run code that reads only from one sensor. Do you have any idea of what might be wrong here? Thanks in advance for your reply,

Best Regards,

Daniel

Apollon77, 2017-08-21:
Should work and also did as I tested it some time ago ... Hm ... No idea at the moment and I also have only one sensor left so I can not test it :-(

lukecyca, 2017-09-04:
@danielfppps I have two sensors on the same I2C bus, and it seem to be working. I have one at 0x20 and the other at 0x21. In the code, I have used two instances of `I2CSoilMoistureSensor`.

You mentioned that both sensors work individually. But do they also work individually when they are both plugged in, but only one is referenced from the code? If not, then you likely have an electrical problem with your I2C bus. Make sure you have strong enough pull-up resistors.

Apollon77, 2017-09-05:
Thank you for your help ! I would close it then

closed 2017-09-05 by Apollon77