Apollon77/I2CSoilMoistureSensor #15: A Wire.requestFrom() should not be followed by a Wire.endTransmission().

Koepel, 2018-03-10:
In the function "readI2CRegister16bitUnsigned()", there is a Wire.endTransmission() after the Wire.requestFrom(). You can remove that second Wire.endTransmission() in that function. 

The Wire.endTransmission should only be used when writing data.

Apollon77, 2018-03-31:
Do you have tested the functionality when you change this? 
I do not have the sensors in use anymore, that's why I need you to test it (or someone else)

Koepel, 2018-03-31:
No, sorry. I was tired of explaining in forums how to use the Wire library. That's why I started to make many issues on Github, trying to tilt it into the right way.
The extra Wire.endTransmission() is not harmless: [Common-mistakes#2](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Common-mistakes#2)

commit d8bce982 (2018-11-05): Update I2CSoilMoistureSensor.cpp

fixes #15
closed 2018-11-05 by Apollon77