Apollon77/I2CSoilMoistureSensor #21: Two sensors

fabio1955, 2021-06-29:
I did try one sensor with the default address (0X20) and it worked. I did try the change address and it worked (0X21). I did try to have two sensors (0X20 + 0X21) and it stopped. Even the I2C scanner dd not work anymore. It is blocked. Then I did try to use the changeSensor method but it didn't work. Then the question. It is possibile to have a two sensor example working? Arduino UNO and Nano tested with 1.8.15 IDE.

kyle-ssg, 2022-08-10:
Any advice on the above? I was considering getting a few of these and looking at the example code this happened to be the only question I had.

kyle-ssg, 2022-08-10:
I'm assuming you tried something along the lines of this @fabio1955 ?

```
I2CSoilMoistureSensor sensor(0x20);
I2CSoilMoistureSensor sensor2(0x21);
```

kyle-ssg, 2022-08-10:
Does anyone have advice on the above?

RT38, 2022-08-11:
I've ran 3+ sensors. I used 0x20, 0x25, 0x30. Low power will cause the
addresses to change so if you batteries are below around 2.5v be ready to
have problems.

You should be able to do a scan and pick up whatever sensors are reporting
and what address they are at if they do change then manually change them
back. Also, run a median filter on the readings if you are turning them off
and back on again. Sometimes the first reading is wrong. Good luck!

marty1982, 2023-03-21:
Hi all!, what about if i have this readings?. It seems that my water chirp sensor is not propperly connected but turned on instead. Weird.

[image]

RT38, 2023-03-21:
Wrong address or a corroded sensor and it's toast. Scan for address to
double check it's correct.

(I haven't touched my sensors in 2+ years so grain of salt)

-Roger

marty1982, 2023-03-21:
Uhm is weird because i have 4 brand new sensors. 

[image]

I have it connected to a Wemos D1.
GND
3V
SCL
SDA

RT38, 2023-03-21:
Check the addresses of your sensors. If you have multiple sensors on one
address you could get those readings.

Make sure all your connections are good. If a loose connection and it gets
low power it can reset address.

-Roger

marty1982, 2023-03-21:
and the weird thing is that previously this worked in a wemos d1 mini with no problem. The wemos d1 mini get frosted and that's because im using another arduino board. I dont't think that could be a problem.